rehosting/penguin #649: add ability to patch (non-binary) files

lockbox, 2025-09-26:
I think it'd be nice from the user perspective to have a config file that does something where you can patch string/config files like these:

**Describe the solution you'd like**
extend `StaticFileAction` with a new variant `string_patch`

```yaml
static_files:
  /etc/init.d/rc.S:
    type: string_patch
    regex: <python regex>
    replace: <string>
```

extend `StaticFileAction` with a new variant `diff_patch`

```yaml
static_files:
  /etc/init.d/rc.S:
    type: diff_patch
    inline: <contents>
    host_patch: <host path> 
```

Would you be open to a contribution implementing this?

zestrada, 2025-09-26:
Sure, we'd welcome a PR!

lockbox, 2025-10-16:
@zestrada is there a mechanism to add unit tests for plugins that don't result in "actions" like the integration tests you guys have?

I'd like to add some unit tests, and some basic integration "did the target get the properly patched file" tests. but afaics the integration tests are more coupled to the "did an action fail and we provide a solution that makes it succeed" paradigm. Is there a recommended (or desired) way to do this?